analogdevicesinc/Pixelpulse2 #201: Build failure on Debian/Jessie

balbirthomas, 2016-06-21:
This package fails to build on Debian Jessie. I am a 
[patch](https://github.com/analogdevicesinc/Pixelpulse2/files/325509/debian-jessie-build.txt) that i used to make the build work. However this meant regressing the Qtquick controls style version and removing the sidebarVisible property from FileDialog as these are not compatible with the Qt version (3.5.2) on Debian Jessie. I wonder if it is possible to make this conditional so that the build works both older and newer GNU/Linux distributions. Support for Debian Jessie (or whatever is Debian Stable) will be much appreciated.

dNechita, 2016-06-22:
Hi,

I looked over your patch and the changes do not break any functionality in Pixelpulse2. Therefore these changes will be added to the master branch or you can make a pull request if you prefer it that way.

Thanks,
-Dan

balbirthomas, 2016-06-22:
Thank you for considering acceptance of patch. If the patch works for you, that is just fine with me. If you absolutely need a pull request, kindly do say so, but it may take me some time to get around to it as I am a bit hard pressed for time now.

dNechita, 2016-06-22:
There is no need for a pull request. I will add your patch.

dNechita, 2016-06-22:
Changes were made with commit: https://github.com/analogdevicesinc/Pixelpulse2/commit/77c854e5f1dfe5334cb2becf2a3fe844037ba762

Thanks!

closed 2016-06-22 by dNechita