analogdevicesinc/Pixelpulse2 #202: usually the voltage range for source voltage is -0.1V to 5.1V

luciansin, 2016-07-21:
Usually you can enter a value lower than 0V(lower limit is -0.1V) and higher than 5V(higher limit is 5.1V) for voltage. When this happens the signals are cutoff.
There are also times when you cannot introduce a value outside this range [0V;5V], but I wasn't able to reproduce what cause that and this good behavior happens rarely. Most often you can set a value outside the range.
[image]

AlexandraTrifan, 2016-07-21:
Fixed with commit 662434be72892fa0f5d15420b27ad9b8c0390ed9.

AlexandraTrifan, 2018-09-05:
Tested on v1.0.4. Can't reproduce it anymore. 
I will close this.

closed 2018-09-05 by AlexandraTrifan