analogdevicesinc/Pixelpulse2 #205: difference between DC RMS and AC RMS

luciansin, 2016-07-26:
There should be a clear difference in pixelpulse plot between DC RMS and AC RMS; what we are displaying now is an AC RMS; you should be able to choose between AC and DC RMS or, for the moment just display AC RMS(with its full name) and display the mean also. 

Lucian

rgetz, 2016-07-26:
I think the terms are "True RMS" and "RMS".

http://en-us.fluke.com/training/training-library/measurements/electricity/what-is-true-rms.html

https://www.tester.co.uk/blog/electrical/multimeter-jargon-busters-true-rms

http://www.escoglobal.com/resources/pdf/white-papers/True_G2.pdf

-Robin

damercer, 2016-07-26:
I agree with Robin ( more or less at least ). True RMS is the best, most exact way to refer to the power level of a signal. It measures the power ( i.e. heat ) delivered to a load, so by definition it is calculated using the entire signal DC and AC parts. Different people ( and sometimes manufacturers ) use shorthand terms like RMS and AC RMS , DC RMS etc. I come down on the side that RMS = True RMS, AC RMS = just the AC portion of signal is used to in RMS formula. DC RMS is a meaningless expression in that power of DC ( V^2/R) is the same as using the RMS formula. 

But to be pedantic if there is some sort of distinction between RMS and True RMS then it follows that there is some sort of False RMS or not True RMS? RMS is RMS True or otherwise. There can be only one!

AlexandraTrifan, 2016-07-27:
Enhancement is now complete. The corresponding commit is 313ee622841c77043444f0d10419800a840479ad.
Right now we are displaying the AC RMS. The label for the RMS value was changed to RMS (AC) and the average is now displayed as well. Is this ok or it should be changed?

closed 2019-07-03 by AlexandraTrifan