analogdevicesinc/Pixelpulse2 #206: Pressing Enter as a requirement to reflect the values in the chart

TonyJ93, 2016-07-28:
Whenever the positive peak, negative peak, and frequency is changed, it is currently required for the user to press enter. If in case the user forgets to press enter, the changed values shall not be reflected to the respective chart. This is in both channel A and B.

Operating System: Windows 7 x64
Pixelpulse2 version: 0.90

dNechita, 2016-08-08:
Fixing this with commit https://github.com/analogdevicesinc/Pixelpulse2/commit/cdfcea348690d8ad681abacf9c8b480113eb5f95
These changes will make the input widgets accept a value when users insert the value without pressing the enter key but only when the widget loses focus.

Thanks,
-Dan

AlexandraTrifan, 2018-09-05:
The value is applied when the widget loses focus or when Enter is pressed on v1.0.4.
Closing this issue.

closed 2018-09-05 by AlexandraTrifan