analogdevicesinc/Pixelpulse2 #209: Auto Launch

damercer, 2016-09-14:
Is there a way to turn off the Auto Launch feature in Pixelpulse2 so that it does not automatically start the program every time the M1K is plugged in?

rgetz, 2016-09-14:
Was thinking about this as well. 

Rather than on/off - we should allow users to pick the tool they want to run. That way they can decide to run PixelPulse, Alice, or MATLAB or nothing. 

- Robin

dNechita, 2016-09-14:
Right now there is no option (listed in Pixelpulse2) to enable/disable it.
However, if it bothers you,  you can close pp2trayer.exe. It has a small icon in the windows icon tray and can be close by right-clicking on it and select Exit. Upon system reboot it will show up again, therefore it is required to remove pp2trayer from the Windows Startup list to prevent it from starting again.

rgetz, 2016-09-14:
Dan:

How hard would be to add this feature to the trayer app?

dNechita, 2016-09-14:
It will be easy to add a new menu item so that when you right-click the tray icon you can deactivate the auto-launching. But it will take a little longer to investigate how to make this option persistent and implement it.

damercer, 2016-09-14:
I almost never use Pilelpulse2, I only loaded it on the ADI laptop as a test. I will probably just un-install Pixelpulse2 for the time being until a version with more options is available.

Thanks

cristi-iacob, 2019-08-20:
Deactivate button added. You can access it by right-clicking the small Pixelpulse Trayer image from the Windows icon tray. Thanks!

closed 2019-08-20 by cristi-iacob