analogdevicesinc/Pixelpulse2 #214: Debian package is missing a few run time dependencies

balbirthomas, 2016-12-08:
Even after installing all the build time (Build-Depends line) and run time (Depends line) listed in the debian/controls file, Pixelpulse2 fails to run as it is missing the following run time dependencies that are not automatically installed when the Debian Pixelpulse2 package is installed. In order to fix this all that is required is add thing the following package as additional run time dependencies in the debian/controls file ("Depends" line of pixelpulse2 package)

         qml-module-qtgraphicaleffects
         qml-module-qtquick-dialogs
         qml-module-qt-labs-folderlistmodel
         qml-module-qt-labs-settings
         qml-module-qtqml-models2

Kindly consider adding these.

rgetz, 2016-12-08:
B:

Thanks for the report - I'm not sure how we missed those - thanks for pointing them out.

Our deb package guru is on vacation, and will be back next week - I will ask him to have a look, or maybe Dan will have time.

Thanks
-Robin

itdaniher, 2018-02-01:
Did anything come of this? Looking to use Pixelpulse2 on Ubuntu 17.10. Thanks!

itdaniher, 2018-02-01:
@balbirthomas what Debian version were you running? These packages don't seem to exist on the latest Ubuntu(s) - perhaps it's a distro-specific packaging issue? In that case, a proper fix might be slightly harder to implement, as right now there's a single buildpath for deb and deb-based distros.

itdaniher, 2018-02-01:
was a partial mismatch: on ubuntu 17.10, the following are required to satisfy the runtime depens:

 * qml-module-qtquick-controls 
 * qml-module-qtquick-layouts 
 * qml-module-qtquick-dialogs
 * qml-module-qt-labs-settings
 * qml-module-qt-labs-settings
 * qml-module-qt-labs-folderlistmodel
 * qml-module-qtqml-models2

balbirthomas, 2018-02-01:
@itdaniher I am using Debian version 9.3 (current stable Debian codename 'stretch'). Only qml-module-qtquick-controls and qml-module-qtquick-layouts seem to be different in the list of build time dependencies that you have reported. And infact I do have both of these installed (and available in Debian 9.3). So there is no harm adding it to the build-depends line.  Also I see that qml-module-qtgraphicaleffects is available in ubuntu from http://packages.ubuntu.com. And I presume you have this installed. 

Indeed if Ubuntu and Debian packages are named/packaged differently this is probably is a more complicated issue than anticipated.  I was able to find some discussion on this at https://raphaelhertzog.com/2010/09/27/different-dependencies-between-debian-and-ubuntu-but-common-source-package/ (See discussion on problems related to different **"build-depends"** in the 'comments' section). I suspect that other package maintainers have faced this issue, and there may be existing official policy and know how on this topic that I am currently unaware of. The Debian developers mailing list may be of some help here. Seems like the easy solution will be to have a meta package that installs all Qt qml modules. Perhaps one may try raising a ticket with both Ubuntu and Debian for this. However adding the collated list of dependencies in this thread should work on both Ubuntu 17.10 and Debian 9 for now.