analogdevicesinc/Pixelpulse2 #218: module "QtQml.Models" is not installed

kaardeco, 2017-07-29:
When I execute the ./pixelpulse2 command, the following error appears:

QQmlApplicationEngine failed to load component
qrc:/qml/main.qml:213 Type DeviceManagerPane unavailable
qrc:/qml/DeviceManagerPane.qml:3 module "QtQml.Models" is not installed

When I run the command "apt-get install qtdeclarative5-models-plugin" this appears:

The package qtdeclarative5-models-plugin  is unavailable, but is referenced by another package.
This may mean that the package is missing, has become obsolete, or
It is only available from another source
However, the following packages replace it:
Qml-module-qtqml-models2: i386 qml-module-qtqml-models2

Could you please help me!

itdaniher, 2017-07-29:
try:

    sudo apt install Qml-module-qtqml-models2

kaardeco, 2017-07-29:
thank u <3 this work!

closed 2017-07-29 by kaardeco