analogdevicesinc/Pixelpulse2 #219: Trouble with Qt5.4 install on zorin/Ubuntu

snoopyreed79, 2018-01-23:
wget http://qtmirror.ics.com/pub/qtproject/development_releases/qt/5.4/5.4.0-rc/qt-opensource-linux-x64-5.4.0-rc.run
chmod 755 qt-o*
./qt-opensource-linux-x64-5.4.0-rc.run

I do the first part with no problem. The last line keeps given me problems I am new at this so I figured I missed something  could someone please help

AlexandraTrifan, 2018-01-23:
The last command should start the Qt installer. What kind of problems do you have? Could you post the error prompted in the terminal?

snoopyreed79, 2018-01-23:
ok i am very new at this i am going through step by step on trying to
install pixelpulse2 the site i am going off is
https://www.element14.com/community/groups/test-and-measurement/blog/2015/02/14/getting-started-with-the-active-learning-module-adalm1000
i have only made it to what i would think of as step 3 in step 3 there are
3 lines of code i get to line 3 and type in what they say to witch is
./qt-opensource-linux-x64-5.4.0-rc.run and i get this bash:
./qt-opensource-linux-x64-5.4.0-rc.run: cannot execute binary file: Exec
format error.
i did what you told me qt-opensource-linux-x64-5.4.0-rc.run and i get this
error qt-opensource-linux-x64-5.4.0-rc.run: command not found
i am sure i am missing something easy but like i said i am new.

thanks for your help
A. Reed

AlexandraTrifan, 2018-01-23:
Could you provide some more information about your Operating system?
After running the 2nd line : chmod 755 qt-o*, could you run the following and post the result?
ls -la | grep qt
uname -a

snoopyreed79, 2018-01-23:
hello
as i have said in the other emails i am new at linux so i looked for a os
that would help me with an easy transition over to ubuntu and i found this
os https://zorinos.com/ i am running there 32bit lite version right now
till i upgrade my system. i ran the 2 line codes and copied below.
again i thank you for the help
A.Reed

snoopy@Snoopys-Laptop:~$ ls -la | grep qt
lrwxrwxrwx  1 root   root          53 Jan 22 04:10 default.conf ->
../../../../share/qtchooser/qt5-x86_64-linux-gnu.conf
-rwxr-xr-x  1 snoopy snoopy  83592874 Jan 22 20:37
qt-opensource-linux-x64-5.4.0-rc.run
-rwxr-xr-x  1 snoopy snoopy 105614506 Jan 22 20:37
qt-opensource-linux-x64-5.4.0-rc.run.1
-rwxr-xr-x  1 snoopy snoopy 568464431 Nov 26  2014
qt-opensource-linux-x64-5.4.0-rc.run.2
snoopy@Snoopys-Laptop:~$

snoopy@Snoopys-Laptop:~$ uname -a
Linux Snoopys-Laptop 4.13.0-26-generic #29~16.04.2-Ubuntu SMP Tue Jan 9
21:38:24 UTC 2018 i686 i686 i686 GNU/Linux
snoopy@Snoopys-Laptop:~$

rgetz, 2018-01-23:
If you aren't familiar with Ubuntu - we provide installers for Windows - why not stick with that?

snoopyreed79, 2018-01-24:
Right now I am working with my sons robotics class and a lot of the
robotic  stuff is on Linux so instead of jumping back and forth or using
wine I am looking at just working with Linux for in the long run things get
easier as you use them.

tfcollins, 2018-01-24:
That QT package requires a 64-bit OS.  Also I doubt anyone has tested PP2 on a 32bit OS since may distros have dropped development for them (even ubuntu).

snoopyreed79, 2018-01-24:
OK I missed that thanks

closed 2018-01-24 by snoopyreed79