analogdevicesinc/Pixelpulse2 #223: Pixelpulse2 broken with firmware 023314a, firmware won't update

itdaniher, 2018-02-01:
I press "Play" and get the following:

[image]

Clicking "Update Firmware" in the device manager pane prints "0: failed to open firmware file," the M1K does not enter bootload mode, and PP2 must be restarted to recognize the device.

[image]

claudiupop, 2018-02-02:
Could you tell me on which operating system are you experiencing this issue ?

Thanks

itdaniher, 2018-02-02:
@claudiupop fully updated Ubuntu 17.10 "artful" on x86_64

AlexandraTrifan, 2018-02-08:
After this commit 59f7df9176cf93323335b07cc25bb447ca33a584 you should be able to update the firmware.

itdaniher, 2018-02-09:
Does the trick. Many thanks!

closed 2018-02-09 by itdaniher