analogdevicesinc/Pixelpulse2 #224: A universal Linux package would be neat.

RustyRaptor, 2018-02-07:
With things like Snappy, Flatpak, and Appimage it's possible to make a universal Linux package that will install on any distro that supports it. Definitely beats having to compile from source. Took me 2 days to get Pixelpulse to compile on Ubuntu 16.04. Too much that can go wrong.

itdaniher, 2018-02-07:
saw this - it's implementation of travis-built app images for a similarly complicated Qt application: https://github.com/miek/inspectrum/pull/143

TheAssassin, 2018-02-08:
@itdaniher with [linuxdeployqt](https://github.com/probonopd/linuxdeployqt) it's pretty easy to create AppImages for Qt and non-Qt apps. Just take a look at the [README](https://github.com/probonopd/linuxdeployqt#using-linuxdeployqt-with-travis-ci). I just combined the template there (with the CMake adaptions listed in a subsection) and the existing Travis config.

The package built there should run on the majority of all Linux distros >= Ubuntu 14.04.

sounddrill31, 2025-03-02:
> With things like Snappy, Flatpak, and Appimage it's possible to make a universal Linux package that will install on any distro that supports it. Definitely beats having to compile from source. Took me 2 days to get Pixelpulse to compile on Ubuntu 16.04. Too much that can go wrong.

I managed to do that!

Link to my builds: https://sounddrill31.github.io/Pixelpulse2/

My build config: https://github.com/pkgforge-dev/Pixelpulse2-AppImage/blob/master/.github/workflows/main.yml#L144-L210