analogdevicesinc/Pixelpulse2 #227: Is there a way I could extend acquisition time in Pixelpulse with Adalm1000 in windows?

rbspg, 2019-05-22:
Hi guys, I am trying to make the acquisition of an analog signal in Adalm1000 with Pixelpulse 2 in Windows, but I could only do it for at maximum 10 seconds. That is the longest option available at Pixelpulse.

At my university there is Matlab that allows me to acquire data up to one minute, but I can't afford to buy it for my notebook... So I was wondering if someone knew a way I could acquire a little bit more input data with Pixelpulse 2. 

Is there any new version that allows me to? Or even a simplified version of Pixelpulse 2 just for the data acquisition (I really don't need graphs as I can make them in excel).

Thanks a lot

kaamil1984, 2020-02-02:
For now - you can try to "Save session" to json file, then edit it, save and restore session.

I changed sampleTime like this:
`    "sampleTime": 100`

I have some glitches but it does work.

AlexandraTrifan, 2020-10-09:
Hi,

Data logging feature was implemented in the latest release. This allows the user to capture data for a longer period of time.
You can find docs for this feature here: https://wiki.analog.com/university/tools/m1k/pixelpulse#data_logging
The latest release is available here: https://github.com/analogdevicesinc/Pixelpulse2/releases/tag/v1.0.5

-Alexandra