analogdevicesinc/Pixelpulse2 #238: Using Pixwelpulse2 to update firmware when m1k is in programming mode

damercer, 2020-06-11:
Hi:

This is somewhat related to the issue I just posted in the Libsmu area. I have an m1k board that is in programming mode ( seen as ADALM1000 SAM-BA WinUSB device in Device Manager. I just installed a fresh copy of Pixelpulse and when I run it with this m1k plugged in I get this screen.

[image]

When I click on the up date firmware button it pauses for a few seconds and when it come back it is still seeing the board as in programming mode, There is no message about power cycling the board to complete the update process. If I click on refresh list before unplugging the board it still reports as being in programming mode. If I power cycle the board with Pixelpulse still running it still see it in programming mode even after clicking on refresh list.

If I close and restart Pixelpulse it now sees the board and is working.

Just thought you might want to make the full process of flashing the firmware more clear and explicit.
Thanks
Doug

cristi-iacob, 2020-06-15:
Hi!

The Pixelpulse version you are using is 1.0.3. In the latest Pixelpulse version (1.0.4) the firmware update feature has changed a little. When having a device in programming mode, after updating the device's firmware, Pixelpulse displays a message to reconnect the device. After disconnecting and connecting back the device (without closing Pixelpulse), Pixelpulse displays its firmware version properly. 

Hope this solves the issue for you!

damercer, 2020-06-15:
I thought I downloaded a fresh copy from the releases?
In Windows the version is showing as 1.0.4

[image]

I'll try again but either way the numbering seems to be off by one?

Doug

cristi-iacob, 2020-06-16:
Hi!

You are right. It seems that the version displayed inside of Pixepulse is wrong. We will solve this in the next release. However, using the latest Pixelpulse version (1.0.4 - the one you have installed) should display a proper message for reconnecting the device in the text field under the Clean log button, after updating its firmware. We haven't yet managed to reproduce this behavior, in which Pixelpulse displays the newly installed firmware only after Pixelpulse's restart. 

We will keep trying to reproduce and solve this issue. Until then, can you please try to reinstall libsmu with the latest official release? If this doesn't solve the issue, you can try installing libsmu from [here](https://ci.appveyor.com/project/analogdevicesinc/libsmu/builds/30613323/job/65ddrt1ued9152yr/artifacts). This is a branch containing some modifications which might have something in common with this issue (some USB transfers fixes). 

Thank you!

damercer, 2020-06-16:
I don't use Pixelpulse at all so I'll wait till the next official release to test this again.
I just wanted to check the firmware update procedure in Pixelpulse to see how well it worked if and when it might need to be used by new students (and faculty) this fall as they roll out remote classes using m1k.

Doug