analogdevicesinc/Pixelpulse2 #239: Disabeling Trayer?

damercer, 2020-06-12:
Hi:

How do I disable the Trayer from starting Pixelpulse EVERY time I plug in an m1k and not uninstall the program? This "feature" is Very annoying.

Thanks

Doug

mhennerich, 2020-06-14:
Hi Doug,

You should be able to disable using the msconfig windows tool.

-Michael

damercer, 2020-06-15:
And what is that, I'm not familiar with it? 

Either way there should be an obvious way to choose during the installation. I don't see that option. There is a choice to not run the trayer upon exit which I unchecked but it still launches each time I plug in the m1k, even after a restart.

cristi-iacob, 2020-06-16:
Hi!

[image]

You can access the Pixelpulse-Trayer icon from the Windows ,,Show hidden icons" menu (the up-arrow from the Windows Taskbar). By right-clicking the icon, you should be able to choose to deactivate its functionality. If the Deactivate option is set, Pixelpulse won't start when a new m1k device is plugged in. 

Hope this solves the issue for you!