analogdevicesinc/Pixelpulse2 #248: Pixelpulse2 windows 10 (1909) crash

ffw8yx, 2020-09-23:
Hi,
i installed on my W10 desktop pixelpulse2 v1.0.5 but after run the program it just crash with no debug info.

info window say: 
Pixelpulse2 has stopped working.

i tried to run it in console, also no debug info.

only what i find out is in system log of windows:

---------------------------------------------------------------------------------
Fault bucket 1841790363512339296, type 5
Event Name: BEX64
Response: Not available
Cab Id: 0

Problem signature:
P1: pixelpulse2.exe
P2: 1.0.5.0
P3: 5f60d421
P4: StackHash_ac46
P5: 0.0.0.0
P6: 00000000
P7: PCH_83_FROM_ntdll+0x000000000009CBC4
P8: c0000005
P9: 0000000000000008
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFA4D.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBA5.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBB6.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBC4.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WERFBE4.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_pixelpulse2.exe_6e15578dcf7ce099e2451e41cc68059b876188_f59a5259_752406d6-797b-415f-9b16-7e778314dde7
------------------------------------------------------------------------------------

this pointing to some paths with "?"  that is strange but im not programmer, im just guy who wana ALDAM1000 use  as analog instrument...

pls help... version 1.0.4 works well....

Best Regards,
Pavel

AlexandraTrifan, 2020-09-24:
Hi,

Could you tell us some more details about your system / machine?
Does Pixelpulse2 crash right after starting the application or after you "Run" the data acquisition? 

Can you open any of the attached files that you listed? If you can, could you attach them here? We need some more details in order to figure out what is wrong here.

What other applications were running on your machine while starting Pixelpulse? I found that this type of issue could appear when there is a conflict between two applications. (we don't know of any conflicts for Pixelpulse2 until now, but it is worth checking)

Could you send a screenshot of the Device Manager while ADALM1000 is connected to the PC? Did the drivers install successfully when Pixelpulse2 was installed? 

Thank you,
-Alexandra

ffw8yx, 2020-09-24:
Hello Alexandra,
i have windows 10 pro, version 1909, osbuild 18363.1082 64bit. I have Lenovo T520 with intel i5-2520M 8G ram... better if you may point to some log file or file that would be enough specific for you.

Pixpulse2 v1.0.5 crash right after starting app. 
[image]
i m trying to attach event log from windows:

[Pixelpulse2_event.zip](https://github.com/analogdevicesinc/Pixelpulse2/files/5278781/Pixelpulse2_event.zip)

 If exist some switch for pixelpulse2 that i may run app in console to get detail debug mode and i have some log out of it please let me know.

i have Alice v1.3 without problem so far. App fall independently on ALDAM1000 connectivity.
[image]
no other special software.... run it as single application, dont count some apps running in bacground that user dont see.....

i had some difficulties with usb driver installation with pixpulse 1.0.5 package, but after i look around and find: https://zadig.akeo.ie/  and than alice has no issue with it...

please let me know what i may do more...
Best Regards,
Pavel

ffw8yx, 2020-09-25:
Hi Alexandra,
today i get other laptop with W10 Enterprise and suprise suprise everything seems to be ok. So it must be related to my laptop and software configuration. I tried to go version back and in version 1.0.4, 3 i start the pixelpulse app sucesfully but i dont see green waveforms, just numbers are running that seems to work...
If i understand well you are using adobe flash player? i what else (libs?)  may be important for graphic interface?
thanks for your help!
Best Regards,
Pavel

AlexandraTrifan, 2020-09-28:
Hello,

It is related to the software configuration, but we want to find out why the package is dependent of software outside of it.
I created a build for you to test, I removed a dependency that might cause the crash you are seeing on the initial laptop.
https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/builds/35404833/artifacts

It is not a signed, official installer, but if you can confirm that it works, then we will update the release.

You can't see the waveforms if you use v1.0.4 or the latest, v1.0.5? 

Thank you,
-Alexandra

ffw8yx, 2020-09-28:
Hi Alexandra,
unfortunately this experiment didnt help, it behaved exactly the same as
v1.0.5, crash during start up.

I cant see waveform in v1.0.4.

V1.0.5 i cannot start at all.

thx for your support... maybe add some logging feature to enable  some
debugging possibilities?

Best Regards,
Pavle

AlexandraTrifan, 2020-09-28:
Hi,

Could you try to run a tool that checks whether all the dependencies are found?
You need to download the 64_Release.zip from here: https://github.com/lucasg/Dependencies/releases/tag/v1.10
You don't need to install anything, just extract the archive. Inside you should find a DependenciesGui.exe. Open the .exe file from that archive and load the Pixelpulse2.exe file (located in C:\Program Files\Analog Devices\Pixelpulse2.

Could you attach here a screenshot of pixelpulse2.exe in DependenciesGui?

Thank you,

-Alexandra

ffw8yx, 2020-09-28:
Hi Alexandra,
i did run it with that special version you prepared and it seems that
dependencies are ok?

[image: image.png]
let me know what else i may do for You.
Best Regards,
Pavel

AlexandraTrifan, 2020-09-28:
Hi, 

I can’t see the attached screenshot.

-Alexandra

ffw8yx, 2020-09-29:
Hi Alexandra,
sorry for my mistake, i send the reply by gmail and evidently it is not able to put here atachement.... so pls have a look:
[image]

also what i noticed, maybe not relevant,  that people who has windows 10 as upgrade from previous version( in my case from windows 7) has some GUI difficulties with some software(i noticed some issue with games)... maybe it is somehow related because the second machine where pixelpulse2 runs fine is clean windows 10 instalation...

thx and have a nice day
Best Regards,
Pavel

AlexandraTrifan, 2020-09-29:
Hi,

The dependencies look good. 
It might be relevant. The upgrade might have some leftovers that cause these difficulties. That is why I asked earlier about some conflicts with other software on the machine.
I can investigate this further and come back with some answers or questions as soon as possible.

Thank you,
-Alexandra

ffw8yx, 2020-09-29:
Hi Alexandra,
it is not experience i have on my own worsktation, its some similarities i
find out with other issues i have seen in discussion forums...
good luck, i will try to find installation iso to get fresh windows 10.
Best Regards,
Pavel

AlexandraTrifan, 2021-03-10:
Hi,

We added some fixes in the latest master version. 
Could you try the new installer and let us know whether you still have the same issues?
 https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/builds/38110683/artifacts

ffw8yx, 2021-03-12:
Hello Alexandra,
i tried to download 64bit portable version and also install version
unfortunately app after successful install did the same as portable. It
open the window( white box appear and windows announce that program stop to
work.
log is pointing to error BEX64:

Version=1
EventType=BEX64
EventTime=132454498848619902
ReportType=2
Consent=1
UploadTime=132454498853974947
ReportStatus=268435456
ReportIdentifier=f7d1665b-262e-444b-a1b0-e4a8b3617b02
IntegratorReportIdentifier=40be706a-9ba6-43b0-8c7a-9c9264687d3b
Wow64Host=34404
NsAppName=pixelpulse2.exe
OriginalFilename=pixelpulse2.exe
AppSessionGuid=00002c8c-0002-0012-1042-dc37aa92d601
TargetAppId=W:000625127e829c6d92e2deae2e8dc05ddaf000000904!0000f522475f2a8472cd71d5e9a775d602ed0ff00fcc!pixelpulse2.exe
TargetAppVer=2020//09//15:14:48:01!65c55c!pixelpulse2.exe
BootId=4294967295
ServiceSplit=2505113784
TargetAsId=1350
UserImpactVector=3145728
IsFatal=1
EtwNonCollectReason=1
Response.BucketId=6e27799dc287cb77598f5a93cfe3fb60
Response.BucketTable=5
Response.LegacyBucketId=1841790363512339296
Response.type=4
Sig[0].Name=Application Name
Sig[0].Value=pixelpulse2.exe
Sig[1].Name=Application Version
Sig[1].Value=1.0.5.0
Sig[2].Name=Application Timestamp
Sig[2].Value=5f60d421
Sig[3].Name=Fault Module Name
Sig[3].Value=StackHash_ac46
Sig[4].Name=Fault Module Version
Sig[4].Value=0.0.0.0
Sig[5].Name=Fault Module Timestamp
Sig[5].Value=00000000
Sig[6].Name=Exception Offset
Sig[6].Value=PCH_83_FROM_ntdll+0x000000000009CBC4
Sig[7].Name=Exception Code
Sig[7].Value=c0000005
Sig[8].Name=Exception Data
Sig[8].Value=0000000000000008
DynamicSig[1].Name=OS Version
DynamicSig[1].Value=10.0.18363.2.0.0.256.48
DynamicSig[2].Name=Locale ID
DynamicSig[2].Value=1033
DynamicSig[22].Name=Additional Information 1
DynamicSig[22].Value=ac46
DynamicSig[23].Name=Additional Information 2
DynamicSig[23].Value=ac462b2fd07cc853b652b5104a1cf381
DynamicSig[24].Name=Additional Information 3
DynamicSig[24].Value=5209
DynamicSig[25].Name=Additional Information 4
DynamicSig[25].Value=52094e25323a51fa58f8da219e9e7e86
UI[2]=C:\Program Files\Analog Devices\Pixelpulse2\pixelpulse2.exe
UI[3]=Pixelpulse2 has stopped working
UI[4]=Windows can check online for a solution to the problem.
UI[5]=Check online for a solution and close the program
UI[6]=Check online for a solution later and close the program
UI[7]=Close the program
LoadedModule[0]=C:\Program Files\Analog Devices\Pixelpulse2\pixelpulse2.exe
LoadedModule[1]=C:\WINDOWS\SYSTEM32\ntdll.dll
[... 227 lines cut ...]
OsInfo[29].Value=10.0.18362.1082.amd64fre.19h1_release.190318-1202
OsInfo[30].Key=buildflightid
OsInfo[30].Value=168e9e15-8814-4f63-9e35-cd92027e45b0
OsInfo[31].Key=edition
OsInfo[31].Value=Professional
OsInfo[32].Key=ring
OsInfo[32].Value=Retail
OsInfo[33].Key=expid
OsInfo[34].Key=containerid
OsInfo[35].Key=containertype
OsInfo[36].Key=edu
OsInfo[36].Value=0
FriendlyEventName=Stopped working
ConsentKey=BEX64
AppName=Pixelpulse2
AppPath=C:\Program Files\Analog Devices\Pixelpulse2\pixelpulse2.exe
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=B350BAB7A082E2F651B57C871B35BE33
MetadataHash=1145041201

i hope it will help,if i can do/try more please let me know.
thx for your support!!,
Best Regards,
Pavel