analogdevicesinc/Pixelpulse2 #249: Pixelpulse2 Driver install failled

gartorant, 2020-11-08:
Hi guys, I've got a problem with the driver on Windows 10 Pro version 1909. 

[image]
[image]

I downloaded the release Pixelpulse2_win_setup.exe version 1.0.5

Thank you for your time.

cristi-iacob, 2020-11-10:
Hi!

We are investigating this issue right now. Until we come up with a solution, can you please answer us to a few questions?
1. Have you tried installing Pixelpulse multiple times? The wizard suggests that if the installation process fails, it might be a good idea to run the installer again. 
2. Did you run the installer using higher permissions for your computer (using run as administrator or run elevated)?
3. Did you have any other version of Pixelpulse or libsmu installed on your PC before running this installer or is this the first time you are using Pixelpulse?

Thank you, 
Cristian

gartorant, 2020-11-10:
Hi Cristian!
1st.- No, it’s the first time but I tried several times installing the Pixelpulse program. 
2nd.- No, I didn’t, sorry. I will try it. 
3rd.- No, I didn’t install Pixelpulse v1.0.5 previously. I could only install Pixelpulse v1.0.4.
P.S.: What is libsmu? I didn’t install the library for M1K because I don’t think I use the Alice’s software.

See you later.

gartorant, 2020-11-23:
Hi Cristian:

1. I tried to install several times. It didn't work. 
2. I tried and the same result. 
3. I installed the Pixelpulse v1.0.4 and it works fine.

jhs-bme, 2021-01-12:
I too have experienced the same problems, attempted the same "fixes" and still no luck.
1. Will this issue be soon resolved?
2. Where can I get a copy of the installer for Pixelpulse v1.0.4?
Thanks

ChrisWattsBSL, 2021-02-02:
I had an existing working installation, not used for some while. I plugged in my ADALM1000 and noticed Bossa Program Port being installed as a USB driver. Tried installing Pixelpulse2 on another machine and got the same problem described here. Bossa Program Port also got installed by windows. Tried removing it with device manager (inc ticking the box to remove software) but still got error message as described above when I tries install again. I'm using Window 7 home edition on the firs machine and 7 pro on the second.

cristi-iacob, 2021-02-05:
Hi! 

We might have found a solution for this issue. We will implement it in the near future and come back to you with the new installer version. 

The problem might come from dpinst.exe, the program which installs the drivers on Windows machines. Dpinst is quite outdated (it is even deleted from the latest Windows 10 versions) and this might cause the issue you encounter. The supported executable for installing the drivers is now pnputil.exe. We will modify the installer to use pnputil and ask you to test it again.

Thank you, 
Cristian

jhs-bme, 2021-02-05:
> This message was sent from a non-IU address. Please exercise caution when clicking links or opening attachments from external sources.

> and come back to you with the new installer version.

Thank you.

=======
John H. Schild, PhD
Associate Professor of Biomedical Engineering
Purdue School of Engineering & Technology
IUPUI

cristi-iacob, 2021-02-08:
Hi!

We implemented the modifications. Can you please try to install Pixelpulse2 using the installer generated by [this build](https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/builds/37664056/artifacts)? 

Please let us know if this installer solves the issue for you.

Thank you,
Cristian

ChrisWattsBSL, 2021-02-10:
I tried the installer from the link above. It fails on both my Win7 pro (32bit) laptop and Win7 home edition (64 bit) desktop. The failure notification from the desktop is attached.
[image]

cristi-iacob, 2021-02-10:
Hi! 

Thank you for your response! We will further investigate this issue and come back to you with another solution.

jhs-bme, 2021-02-12:
Hi Cristian
Sorry for the delayed response but I wanted to have time to work w/a clean install on a new machine: OS: Windows 10, 64 bit

Simply ran the installer you provided: Pixelpulse2_win_setup.exe

The package seemed to install OK but in using Pixelpulse2 to test out the ADALM1000 there are some “oddities” in using the software all relative to using the board which I connect to a USB port before starting Pixelpulse2.

  1.  The power light is supposed to go from red to green but this does not occur right away and sometimes not at all
  2.  It almost seems like the mouse clicks are sometimes being recognized and sometimes not (new mouse, default setting, etc.)
  3.  After fiddling w/the software, changing the SMU and using different waveforms I can get the setup/output I want.

I know the behavior seems strange but it seems like sometimes it responds to mouse clicks as expected and sometime it gives no response.

=======
John H. Schild, PhD
Associate Professor
Biomedical Engineering
IUPUI

cristi-iacob, 2021-02-15:
Hi! 

We made some further investigations on this issue and created a new [installer](https://ci.appveyor.com/project/analogdevicesinc/pixelpulse2/builds/37775542/artifacts). In this version we updated the driver installation process. Can you please check if this version solves the issues for you? 

@jhs-bme the first problem is most probably related to the fact that the driver installation was not working properly. The mouse clicking issue is a little weirder indeed, but I suspect to have the same origins as the first one (sometimes the callbacks to the device are working properly, but sometimes not - and this thing is probably hidden by the GUI elements from Pixelpulse). There were some problems with the old dpinst executable we were using for the driver installation. We solved these issues in this installer. 

@ChrisWattsBSL we removed the usage of the pnputil executable (the one which was causing the issues for you).

Please let us know if this installer solves the issues for you! Thank you!

ChrisWattsBSL, 2021-02-17:
Hi.
That installer works on my laptop (32bit win7), not tried the desktop (64bit win7) yet. The problem I have now is:
[image]
I have tried this at home and at work, and get the same result. Other internet stuff working fine on the laptop.
Any ideas?

cristi-iacob, 2021-02-17:
Hi @ChrisWattsBSL! 

I'm glad to hear that the installer works for you! 

About the firmware update issue: I tested this feature just now and it seems to work properly. I have also checked the URL Pixelpulse is trying to access in order to get information about the latest firmware version and it is working and contains all the requested information. I suspect that the issue might come from your network/firewall configuration. Probably one of your settings blocks Pixelpulse from sending requests to [the page that contains the information about the firmware version](https://api.github.com/repos/analogdevicesinc/m1k-fw/releases). Depending on your configurations, you might consider disabling the option that blocks Pixelpulse. 
If you don't want to do this, you can manually update the firmware using the smu executable provided by libsmu. You can read more about this process [here](https://wiki.analog.com/university/tools/m1k/firmware-upgrade#update_using_the_smu_executable). 

Thank you for your response and please let us know if you have any further issues!

ChrisWattsBSL, 2021-02-18:
The installer works on the desktop machine (win 7 64 bit) too. I had to disable AVG for this to happen. The laptop has Norton on it that did not get in the way of the install. I tried disabling the firewall on the laptop but the firmware update issue persists. It is present on the desktop as well. This is now just academic as the smu utility works just fine and I now have a working M1K with the latest firmware. Many thanks for your help.
Chris.

AlexandraTrifan, 2021-03-10:
The fixes were merged to master ( https://github.com/analogdevicesinc/Pixelpulse2/pull/253 ).
The driver should install properly now and a new version will be released soon with these fixes.
Will close this issue, can be reopened if more problems appear.

closed 2021-03-10 by AlexandraTrifan

golikoje-prog, 2025-10-16:
I am facing the same issue with which this topic started, but I couldn't get the final thing clearly that what should I ultimately do? 
Windows 11
I am using ADALM1000 .. 
When I connected it, it could detect the kit, but ig was not showing properly how to give the voltage or current inputs... Or may be I am not able to figure it out..?