analogdevicesinc/Pixelpulse2 #252: Pixelpulse2 - not building from source

76nic, 2021-01-31:
Dear AD,
I have been following your instructions to build Pixelpulse2 from source in Ubuntu 20.04. It does not build! There might be a mistake in the libusb version? I get the following error:

-- Checking for module 'libusb-1.0'
--  Found libusb-1.0, version 1.0.23
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files
LIBSMU_INCLUDE_DIRS
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
   used as include directory in directory /home/ku/development/pixelpulse2
LIBSMU_LIBRARIES
    linked by target "pixelpulse2" in directory /home/ku/development/pixelpulse2

-- Configuring incomplete, errors occurred!

Many thanks for your help!
Best regards,
76nic

AlexandraTrifan, 2021-02-01:
Did you build and install libsmu?

76nic, 2021-02-01:
Many thanks Alexandra for your quick reply and the hint regarding libsmu.
I was following some old instruction https://wiki.analog.com/university/tools/m1k/pixelpulse/build that are liked from your "main page" https://wiki.analog.com/university/tools/m1k/pixelpulse.

Now I was following the instruction on that page: https://github.com/analogdevicesinc/Pixelpulse2
The libsmu deb package installed just fine.
However, the Pixelpulse2 build results in an error most likely related to the cmake command! Cmake is not installed on Ubuntu 20.4, so I installed the latest version 3.16.3-1ubuntu. Is there a specific version of cmake that is needed to build this (older compiler)?
Many thanks!
Best regards,
76nic

Error:
ku@KuVM:~/pixelpulse2/build$ cmake .. 
-- The CXX compiler identification is unknown 
-- The C compiler identification is GNU 9.3.0 
CMake Error at CMakeLists.txt:7 (project): 
 No CMAKE_CXX_COMPILER could be found. 

 Tell CMake where to find the compiler by setting either the environment 
 variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path 
 to the compiler, or to the compiler name if it is in the PATH. 
-- Check for working C compiler: /usr/bin/cc 
-- Check for working C compiler: /usr/bin/cc -- works 
-- Detecting C compiler ABI info 
-- Detecting C compiler ABI info - done 
-- Detecting C compile features 
-- Detecting C compile features - done 
-- Configuring incomplete, errors occurred!

76nic, 2021-02-03:
I was able to build it! Again many thanks!
Best regards,
76nic

closed 2021-02-03 by 76nic