analogdevicesinc/Pixelpulse2 #256: Segmentation Fault on Ubuntu 20.04

meschi, 2021-10-03:
I've compiled pixelpulse2 according to Readme.md and installed libsmu-1.0.3-ubuntu-20.04-amd64.deb because compilation on my machine fails.
When I'm starting the program the GUI appears for a short time, then the software crashes with a segmentation fault. This is the stdout:

qrc:/qml/DeviceManagerPane.qml:159:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/DeviceManagerPane.qml:357:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/DeviceManagerPane.qml:419:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/ContentPane.qml:60:3: QML MouseArea: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/ColorControlDialog.qml:25:9: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:83:21: QML TimelineHeader: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:126:17: QML Item: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:83:21: QML TimelineHeader: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:83:21: QML TimelineHeader: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:126:17: QML Item: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:126:17: QML Item: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Segmentation fault (core dumped)

meschi, 2021-10-04:
Today I noticed that pixelpulse2 works with a version od libsmu I compiled myself. After some thought it seems that might be caused by plugging the adalm1000 out and in again. If this is true it might be a good idea to add a note to the installation instructions.

I had plugged in the board before installing and compiling the software. I encountered the segmentation fault problems before plugging the board out and in again.

AlexandraTrifan, 2021-10-04:
Hi,

We will look into reproducing this, check out what's causing it and come back with some answers as soon as possible.

Thank you!
-Alexandra

Ewpratten, 2022-04-23:
Exact same issue here on fedora 35 after compiling master branches of smu and pixelpulse myself

vishal-ahirwar, 2024-01-26:
yeah I'm also facing same issue