analogdevicesinc/Pixelpulse2 #260: Erratic mouse in Mac OS Catalina

peterkinget, 2022-11-12:
Hi, 

I am trying to use the ADALM 1000 on a Macbook Pro 13 inch 2020. I have Mac OS Catalina 10.15.7. I installed pixelpulse and it starts but the mouse behaves strangly. 

When you click somewhere you get a repeat of the action related to the previous click somewhere else on the screen ... 

I am attaching a screen recording to illustrate the issue. You do not see the actual clicking, but I tried to click on different places in the screen and I do get a response as if I am clicking in a different region ... 

https://user-images.githubusercontent.com/46092360/201484818-9941e227-c6ca-4c69-a22d-877abd9d98e9.mov

Best -- Peter Kinget

damercer, 2022-11-14:
@AlexandraTrifan this seems to also be happening in Windows. From About: 
Built: 2020/09/15    Version: 1.0.5-g943a506
The most recent release is v1.0.5, published at Wed Sep 16 04:30:03 2020 GMT-0400.
It is available for download at https://github.com/analogdevicesinc/Pixelpulse2/releases/tag/v1.0.5.

When I left click on one of the buttons like the gear or play etc. The program stays in that function and when ever I left click the function repeats. If I right click twice it clears this stuck state until I do another left click and again it get stuck.

Obviously we don't have a lot of new pixelpulse users since 2020/09/15 or this would have come up sooner.

Thanks doug

Houseman74, 2023-04-24:
Same happens to me. That's why I landed here. Mac OS Mojave. also I noticed that the whole fonts are blurred. Seems no support is given to this kind of software....