analogdevicesinc/Pixelpulse2 #266: pixelpulse2 crashing after sort period of time on linux

vishal-ahirwar, 2024-01-18:
pixelpulse2 crashing after sort period of time on linux
distro: popOS 22.04LTS
qt version : 5.15 LTS
I followed exact given instruction   
Log : 
`qrc:/qml/DeviceManagerPane.qml:347:5: QML Connections: Implicitly defined onFoo properties in Connections are deprecated. Use this syntax instead: function onFoo(<arguments>) { ... }
qrc:/qml/DeviceManagerPane.qml:159:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/DeviceManagerPane.qml:357:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/DeviceManagerPane.qml:419:3: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/ContentPane.qml:60:3: QML MouseArea: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/ColorControlDialog.qml:25:9: QML Rectangle: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:85:21: QML TimelineHeader: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:128:17: QML Item: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:85:21: QML TimelineHeader: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
qrc:/qml/main.qml:128:17: QML Item: Detected anchors on an item that is managed by a layout. This is undefined behavior; use Layout.alignment instead.
Segmentation fault (core dumped)
`

sounddrill31, 2024-12-13:
Out of curiosity - Does this also happen when running without a device attached?

vishal-ahirwar, 2024-12-27:
Ohh sry I totally forgot about it was actually crashing only on fedora , it was working just fine on Debian based distros